stianeikeland/go-rpio #7: Errors on go get

danhardman, 2016-01-14:
I get the following output when I get this package.

    go get github.com/stianeikeland/go-rpio
    # github.com/stianeikeland/go-rpio
    ..\..\..\github.com\stianeikeland\go-rpio\rpio.go:290: undefined: syscall.Mmap
    ..\..\..\github.com\stianeikeland\go-rpio\rpio.go:294: undefined: syscall.PROT_READ
    ..\..\..\github.com\stianeikeland\go-rpio\rpio.go:294: undefined: syscall.PROT_WRITE
    ..\..\..\github.com\stianeikeland\go-rpio\rpio.go:295: undefined: syscall.MAP_SHARED
    ..\..\..\github.com\stianeikeland\go-rpio\rpio.go:315: undefined: syscall.Munmap

closed 2016-01-17 by danhardman

z26100, 2016-02-20:
Try to change goos / gooarch to linux / arm.

danhardman, 2016-02-20:
That was the problem... Feel like this should be explained in the syscall package for when noobs like me come around!