stianeikeland/go-rpio #8: Input interrupts

brian-armstrong, 2016-01-28:
Hi,

Have you considered adding input interrupts? You can get them from the files exposed by sysfs in /sys/class/gpio. I've made a library which does this at https://github.com/brian-armstrong/gpio

sysfs can also expose i2c and spi if you uncomment the modules in /boot/config.txt and then reboot. This might be an easy way for you to add i2c and spi functionality as well.

curiositry, 2017-03-21:
@stianeikeland Is this on the roadmap … or should I find a different GPIO library?

brian-armstrong, 2017-03-21:
@curiositry Try mine, it's linked in my comment :)

curiositry, 2017-03-21:
@brian-armstrong Thanks, I’ll take a look! I’ve already built this library into my app, though, and I’m hoping I won’t have to abandon it…

stianeikeland, 2017-03-21:
I'm afraid I don't have time at the moment, but if anyone is willing then a PR is welcome :)

brian-armstrong, 2017-03-22:
@curiositry If you find my gpio lib lacking anything just let me know :)

closed 2017-12-11 by brian-armstrong