stianeikeland/go-rpio #9: A way to set inbetween low and high

jwhitcraft, 2016-02-12:
I'm working with an RGB led strip and i'd like to be able to control the brightness via the gpio, but with this lib i only get High or Low, no way to do it in between.

Just wondering if it's possible?

Thanks.

mihaipora, 2016-02-12:
It is possible using PWM (Pulse Width Modulation).
Arduino libraries have a function called analogWrite which actually generates a square wave with a frequency of 490 or 980 Hz. 
The duty cycle represents the analog value.

Read here about analogWrite:
https://www.arduino.cc/en/Reference/AnalogWrite
And sample application for controlling LED intensity using PWM which explains also how the duty cycle drives the intensity:
http://www.instructables.com/id/Using-PWM-to-control-the-light-intensity-of-a-LED/

Another good explanation and some hints for Raspberry Pi here:
https://learn.adafruit.com/adafruit-raspberry-pi-lesson-9-controlling-a-dc-motor/pwm

phedoreanu, 2016-04-09:
@jwhitcraft Try https://github.com/sarfata/pi-blaster. It creates `/dev/pi-blaster` and then you can do:
```sh
echo "17=0.2" > /dev/pi-blaster
```
to set GPIO pin 17 to a PWM of 20%.

closed 2019-12-03 by jwhitcraft