stianeikeland/go-rpio #25: Raspberry Pi Zero W compatibility issue

mikumaycry, 2018-07-21:
Tested on Raspberry Pi Zero W
**OS Version:** RASPBIAN STRETCH LITE (Release Date: 2018-06-27)
**Kernel Version:** Linux raspberrypi 4.14.52+ #1123 Wed Jun 27 17:05:32 BST 2018 armv6l GNU/Linux
**Go Version:** go1.10.3 linux/arm
**Problems:** I connect to pi via ssh, system freeze after initialize GPIO, but the same code works well on Raspberry Pi 3 which has the same OS Version, Kernel Version and Go Version. 
**sample code:**
```
const (
	io1    = uint8(5)
	rst    = uint8(13)
	intIn  = uint(26)
	intOut = uint(21)
)
var outPin rpio.Pin
var inPin rpio.Pin
func startGPIO() {
	err := rpio.Open()
	if err != nil {
		log.Fatal(err)
	}
	log.Info("open rpio successfully")
	io1Pin := rpio.Pin(io1)
	io1Pin.Output()
	io1Pin.Low()

	rstPin := rpio.Pin(rst)
	rstPin.Output()
	rstPin.High()

	outPin = rpio.Pin(intOut)
	outPin.Input()
	outPin.Detect(rpio.FallEdge)

	inPin = rpio.Pin(intIn)
	inPin.Output()
	inPin.High()
}
```

stianeikeland, 2018-07-24:
I don't have any Pi Zeros I'm afraid, so not sure what's going on here.

Have you tried different pins? Maybe the mapping is different and 5,13,26 or 21 is used for something critical?

Anyone else using this library with a Pi Zero W here?

mikumaycry, 2018-07-26:
Pi Zero W has the same Pin layout with Pi 3B, I try to change pins but still didn't work.
But another library which does GPIO on systems with /sys/class/gpio (sysfs) works on Pi Zero w.
Link: https://github.com/brian-armstrong/gpio

mikumaycry, 2018-07-28:
I think the problem may be the getBase func

**Sample Code**
```
package main

import (
	"bytes"
	"encoding/binary"
	"fmt"
	"os"
)

func main() {
	fmt.Print(getBase())
}

func getBase() (base int64) {
	ranges, err := os.Open("/proc/device-tree/soc/ranges")
	defer ranges.Close()
	if err != nil {
		return
	}
	b := make([]byte, 4)
	n, err := ranges.ReadAt(b, 4)
	if n != 4 || err != nil {
		return
	}
	buf := bytes.NewReader(b)
	var out uint32
	err = binary.Read(buf, binary.BigEndian, &out)
	if err != nil {
		return
	}
	return int64(out)
}
```
**Result**

```
GOOS=linux GOARCH=arm go build -ldflags "-s -w" -o getBase main.go
sudo ./getBase
[1]    669 illegal hardware instruction  sudo ./getBase
```
The library failed to open **/proc/device-tree/soc/ranges** and the error is not properly handled. The **RPI.GPIO** library also tries to open **/proc/device-tree/soc/ranges**, but use hard coded base if error occured
Sample Code ( from c_gpio.c setup func )
```
#define BCM2708_PERI_BASE_DEFAULT   0x20000000
#define BCM2709_PERI_BASE_DEFAULT   0x3f000000

// determine peri_base
    if ((fp = fopen("rranges", "rb")) != NULL) {
        // get peri base from device tree
        fseek(fp, 4, SEEK_SET);
        if (fread(buf, 1, sizeof buf, fp) == sizeof buf) {
            peri_base = buf[0] << 24 | buf[1] << 16 | buf[2] << 8 | buf[3] << 0;
        }
        fclose(fp);
    } else {
        // guess peri base based on /proc/cpuinfo hardware field
        if ((fp = fopen("/proc/cpuinfo", "r")) == NULL)
            return SETUP_CPUINFO_FAIL;

        while(!feof(fp) && !found) {
            fgets(buffer, sizeof(buffer), fp);
            sscanf(buffer, "Hardware	: %s", hardware);
            if (strcmp(hardware, "BCM2708") == 0 || strcmp(hardware, "BCM2835") == 0) {
                // pi 1 hardware
                peri_base = BCM2708_PERI_BASE_DEFAULT;
                found = 1;
            } else if (strcmp(hardware, "BCM2709") == 0 || strcmp(hardware, "BCM2836") == 0) {
                // pi 2 hardware
                peri_base = BCM2709_PERI_BASE_DEFAULT;
                found = 1;
            }
        }
        fclose(fp);
        if (!found)
            return SETUP_NOT_RPI_FAIL;
    }
```

b-x, 2018-08-01:
I can confirm this issue with Raspberry Pi Zero W.
My Pi freezes after pin.Detect().

Drahoslav7, 2018-08-02:
It works fine with my Pi Zero W, but I have a different environment.

**OS:**  Raspbian GNU/Linux 8.0 (jessie)
**Kernel:**  Linux zero 4.9.35+ #1014 Fri Jun 30 14:34:49 BST 2017 armv6l GNU/Linux
**Go:** go1.8.3 linux/arm

Drahoslav7, 2018-08-02:
@mikumaycry You might hit the nail on the head with the `getBase`.
Think is `go-rpio` always fallback to hardcoded adress `0x20000000` (BCM283**5**)

Rpi Zero probably uses  `0x3f000000` (BCM283**6**). Hence it does not work.

I can make fix inspired by the code you posted. But I still see `BCM2835` in output of  `cat /cpu/info`, instead of `BCM2836`, how is that?

Anyway, the real question is why the file `/proc/device-tree/soc/ranges` can not be opened.

Drahoslav7, 2018-08-02:
@mikumaycry BTW, you forgot the fallback line https://github.com/stianeikeland/go-rpio/blob/master/rpio.go#L628 in your sample code.

mikumaycry, 2018-08-03:
@Drahoslav7 
BCM2835 has default GPIO base and it won't use 0x3f000000. Details can be found here:
https://www.raspberrypi.org/documentation/hardware/raspberrypi/bcm2835/BCM2835-ARM-Peripherals.pdf
@stianeikeland 
But I think I've located the problem.
The default GOARM is 7 on macOS, the same as Raspberry Pi 2, 3. But default GOARM is 6 on Raspberry Pi A, A+, B, B+, Zero.
Before I upgrade RPi Zero kernel to latest, it's almost impossible to compile code on it, the go build command usually never stops.
I cross compile on mac and then copy executable file to RPi Zero, so the programs stuck due to wrong GOARM. 
I just recompiled sample code on RPi Zero and it successfully open ```/proc/device-tree/soc/ranges``` and return.

stianeikeland, 2018-08-03:
Goodie! So if you try using `GOARM=6` on your mac, it works?

If that's the case, we should probably make a note of it in the readme.md

closed 2018-08-04 by mikumaycry