stianeikeland/go-rpio #27: Go mod not getting the latest version

xiegeo, 2018-08-28:
Using the mod feature of newly released go1.11 and importing go-rpio as a library causes a downgrade to `v3.0.0`. 

In `go.mod` file, the version is listed as `github.com/stianeikeland/go-rpio v3.0.0+incompatible`, where: 
- `v3.0.0` is the larges version tagged following the [SIV](https://research.swtch.com/vgo-import), since `4.0.0` and `4.1.0` are missing `v` so they are ignored.
- `+incompatible` prevent the import path from adding the major version.

Adding the tags `v4.0.0` and `v4.1.0` should fix the immediate problem. 

[more information about +incompatible flag form another Go project.](https://github.com/go-chi/chi/issues/302#issuecomment-407851258)

stianeikeland, 2018-11-29:
Hey, thanks, could you try again now? `v4.3.0` was just released and tagged.

stianeikeland, 2018-11-30:
Thanks man, think I got it working now, added /v4 and published a new release. Seems to be working on my end.

closed 2018-11-30 by stianeikeland