stianeikeland/go-rpio #34: [Feature request / clarification] servo pulse width

Journerist, 2018-11-04:
Hey everyone,

first, thank you very much for creating this library. Right now, I try to build an autonomous drone and I would love use go-lang (I am a beginner in electronics and go-lang) for learning. Similar libraries written in in nodejs and python usually just use the c-library. They offer a mehod called setPulseWidth that uses set_servo_pulsewidth in C. Are there any plans to implement it in go-rpio?

Right now, I don't feel like I can help. But maybe this issue is also a good way to clearify things. In theory, I read that pulse width, duty cycle and frequency are related. Actually, it should be possible to use casual PWM in combination with duty cycle and frequency or am I wrong?

I just tried it, set the frequency to 50 and set a duty cycle like 50 % without success. Maybe you have an idea where to start or even better, how I can use the existing implementation to mimic the (servo-)pulseWidth implementation of other libraries.

Thanks in advance

(Keep up the good work!)

Drahoslav7, 2018-11-05:
Also, you can use [pi-blaster](https://github.com/sarfata/pi-blaster) with go - can be used on any pin.

```go
file, err := os.Open("/dev/pi-blaster")
...
file.WriteString("17=0.5") // will set pin 17 to 50% pwm
```
disadvantage is worse response time, which might not be sutable for your use case.

Also, you might what to check this https://gobot.io - they have limited support for raspberry pi, but in case you decide to switch to another platform, this might be the way to go.

Journerist, 2018-11-06:
Alright, thank you for the insights. Especially the part that this library is not able to control 4 independent pins saved me a lot of time I think :)).

My fun project is about building an autonomous indoor drone. Therefore, a decent performance is required but I think not as much as a racing drone would require. Basically, the impulses need to be good enough to control the electronic speech controle that controls the brushless motors. Balancing and other concurrent tasks should be fulfilllable by checking the acceleration sensors regularly. But I will see... Maybe I take my time and build a simple balancing script with different languages and libraries to compare these performance wise.

First, I will play around with the pi-blaster using go. There is a go lang library that uses the command line interface that seems promising. If this does not performance well then I will try to use cgo.

Thank you for all the advices! Keep up the good work!

closed 2018-11-06 by Journerist