stianeikeland/go-rpio #35: EdgeDetection crashes Raspberry Pi 3

eyJhb, 2018-11-26:
Using `Detect` and running a very basic program on the Raspberry Pi 3, will result in the whole Raspberry Pi locking up (and needing a forced restart).
This only happens on a pin with very many rises/falls, currently I am using this for a RF receiver, so there are quite a lot of rises/falls.

The code I am using is this ...

```
package main

import (
	"fmt"
	"time"

	rpio "github.com/stianeikeland/go-rpio"
)

// This program is just going to be
// a POC of the sender working.
// This should and WILL be cleaned up
// I do not care, I will not fucking
// write the report without having
// the time to do this properly

// Raspberry Pi 3 GPIO
// https://i.imgur.com/PPFf2ul.png
const rx_pin_nr = 2

var preamble = [8]int{1, 0, 1, 0, 1, 0, 1, 0}
var sfd = [8]int{1, 0, 1, 0, 1, 0, 1, 1}
var rx_pin rpio.Pin

func main() {
	// initialize go-rpio
	err := rpio.Open()

	if err != nil {
		fmt.Println(err)
	}

	// set our pin as a output
	rx_pin = rpio.Pin(rx_pin_nr)
	rx_pin.Input()
	rx_pin.Detect(rpio.RiseEdge)

	for {
		// sendData(test)
		fmt.Println("Running stuff")
		time.Sleep(100 * time.Millisecond)
	}
}

func check_sync() {

}
```

Not even doing anything with it. I don't know what the solution would be, but I think that we could get the interrupt etc. if we used such a system as epoll (http://man7.org/linux/man-pages/man7/epoll.7.html), I know it is uses for another Go system for GPIO interaction here - https://godoc.org/github.com/kidoman/embd

eyJhb, 2018-11-26:
Edit, it is done using this - https://golang.org/pkg/syscall/#EpollCreate1

Drahoslav7, 2018-11-27:
Hi,
this seems to be the same issue as in #33, (which I am not able to reproduce, unfortunately). Does it freeze every time as soon as `Detect` method is called, or only when you connect it to RF receiver and generate some edges?

Could you provide me more info about your environment?

eyJhb, 2018-11-27:
1. Using the newest version of raspbian downloaded from raspberrypi.org three days ago (and ran update/upgrade)
2. Going a direct `go run main.go` when running my code.

But I will check number 1 tomorrow to make sure! :)

eyJhb, 2018-11-28:
# Setup
- Raspberry Pi 3 model B v1.2
- RF sender/receiver [here](https://eckstein-shop.de/433Mhz-RF-Wireless-Sender-Empfaenger-Modul-Transmitter-Receiver-Kit-Arduino)
- Power supply for RPi [here](https://uk.rs-online.com/web/p/ac-dc-adapters/9098126/) - sometimes relevant for issues on the RPi..

Running the latest version of Raspbian (2018-11-13-raspbian-stretch-lite), having the latests updates using `sudo apt update && sudo apt upgrade -y`

```
pi@hub:~ $ uname -a
Linux hub 4.14.79-v7+ #1159 SMP Sun Nov 4 17:50:20 GMT 2018 armv7l GNU/Linux
pi@hub:~ $ lsb_release -a
No LSB modules are available.
Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 9.6 (stretch)
Release:	9.6
Codename:	stretch
pi@hub:~ $ 
```

# Code execution
Running the code below WITHOUT the RF receiver attached, prints "works" and then exists the program and everything works fine.
But attaching the RF receiver, it will print out works and then just completely freeze up and I cannot ping the host, nmap it, ssh into it or anything. I need to unplug the power and plug it in again.

Currently I am just using the polling method, and checking the state each time.
There are not that many rises and falls, but it does of course still have some jitter sometimes.

`main.go`
```
package main

import (
	"fmt"
	"time"

	rpio "github.com/stianeikeland/go-rpio"
)

func main() {
	// initialize go-rpio
	err := rpio.Open()

	if err != nil {
		fmt.Println(err)
	}

	// set our pin as a output
	rx_pin := rpio.Pin(2)
	rx_pin.Input()
	rx_pin.Detect(rpio.RiseEdge)
	fmt.Println("Works..")
}
```

# Output from hexdump
```
00000000  7e 00 00 00 3f 00 00 00  01 00 00 00 40 00 00 00  |~...?.......@...|
00000010  40 00 00 00 00 00 10 00                           |@.......|
00000018
```

# Final notes
If you lived nearby, I would be able to give you a module for testing (if the university agrees), but doesn't look like it.

Drahoslav7, 2018-11-29:
That is really some kind of magic.

Maybe pi just can't handle faster changes.
You can edit the test file and use pin in `Clock` mode, and gradually increase the frequency to see if there is problem in that. Actually I can do this myself.

Also we can try to use "asynchronous" GPAREN/GPAFEN registers for detection instead of "synchronous" GPREN/GPFEN registers. The difference is the synchronous looking for 011/100 patterns, and the other dont - so they should be able to detect even shorter pulses. Just change lines 351/352 in `rpio.go` to
```go
renReg := p/32 + 31
fenReg := p/32 + 34
```
See chapter 6.1 in https://www.raspberrypi.org/app/uploads/2012/02/BCM2835-ARM-Peripherals.pdf if you are interested in details.

roffe, 2018-11-30:
@Drahoslav7 awesome, i'll have results in a few minutes after testing `dtoverlay=gpio-no-irq`

roffe, 2018-11-30:
@Drahoslav7 my code works like a charm the dtoverlay setting, thank you

Drahoslav7, 2018-11-30:
I'm happy we figured it out.
Hope it works for @eyJhb as well. Feel free to reopen this if not.

closed 2018-11-30 by Drahoslav7

eyJhb, 2018-11-30:
@Drahoslav7 haven't had time to test it, currently in the mid of getting all the code to work together before the deadline.
Will test it out :) - But wouldn't it be awesome, to have actual interrupts?

wjessop, 2023-09-06:
I had this issue on my Pi 4B running the program as root. Adding `dtoverlay=gpio-no-irq` to the boot config fixed it. Using go-rpio v4.6.0.