stianeikeland/go-rpio #38: Corrupts gpiomem as non-root

yath, 2018-12-10:
The following code from rpio.Open implies that /dev/mem and /dev/gpiomem are equivalent, the latter only ignores certain writes:
```
	// Open fd for rw mem access; try dev/mem first (need root)
	file, err = os.OpenFile("/dev/mem", os.O_RDWR|os.O_SYNC, 0)
	if os.IsPermission(err) { // try gpiomem otherwise (some extra functions like clock and pwm setting wont work)
		file, err = os.OpenFile("/dev/gpiomem", os.O_RDWR|os.O_SYNC, 0)
	}
```

However judging from http://www.macs.hw.ac.uk/~hwloidl/hackspace/linux/drivers/char/broadcom/bcm2835-gpiomem.c, /dev/gpiomem *only* maps the GPIO page.

The following program, running as non-root only, crashes a Raspberry 3 Model B+ latest at the first attempt to set the duty cycle. Crashing means “not reachable via IP anymore” in my case, presumably because random GPIO memory was overwritten.

```
package main

import (
        "fmt"
        rpio "github.com/stianeikeland/go-rpio"
        "time"
)

func main() {
        rpio.Open()
        defer rpio.Close()

        p := rpio.Pin(12)
        p.Mode(rpio.Pwm)
        p.Freq(4688 * 0xFFFF)
        for i := 0; i <= int(^uint16(0)); i++ {
                fmt.Println("setting", i)
                p.DutyCycle(uint32(i), 0xffff)
                time.Sleep(10 * time.Millisecond)
        }
}
```

stianeikeland, 2018-12-20:
@Drahoslav7 - Do you think this and #39 could be related?

Drahoslav7, 2018-12-20:
@stianeikeland 
It seems to be the same issue.
I think it might be caused by some recent changes in the current Raspbian/kernel, because it wasn't a problem before.

Unfortunately, I don't have time right now to investigate further.

A temporal solution might be to add check if opened file is `/dev/gpiomem` and only call `memMap` for `gpioMem` and not for the other `***Mem` slices in this case.

Can somebody try it?

@yath Does your Pi start normally after reset or is the Rasbian corrupted? Or the hardware is? - hope it is not. Shouldn't be possible without root to do any real demage, right?