stianeikeland/go-rpio #39: Rpi no boot after pwm

Karolcz125, 2018-12-19:
I've launched code on my raspberry pi 3b+ which is simillar to this https://github.com/stianeikeland/go-rpio/blob/master/examples/pwm/pwm.go
on my raspberry pi 3b+
```
func main() {
        err := rpio.Open()
        if err != nil {
                os.Exit(1)
        }
        defer rpio.Close()

        pin := rpio.Pin(7)
        pin.Mode(rpio.Pwm)
        pin.Freq(60*60)
        pin.DutyCycle(0, 60)
        // the LED will be blinking at 60Hz
        // (source frequency divided by cycle length => 60*60/60 = 2000)
}
```
iirc it was exactly like this. After running it ssh freezed and after restarting raspberry, wont boot again. Do you know what could be the problem?

Karolcz125, 2018-12-20:
I've done everything you said except for creating empty file ssh on SD card which I don't get. Tried booting from different card (which is new), as well as booting from USB stick.
Nothing from [this list](https://elinux.org/R-Pi_Troubleshooting#Red_power_LED_is_on.2C_green_LED_does_not_flash.2C_nothing_on_display) is won't work too. 

PSU itself should be okay, as it's solid 5V/2A, I've checked a couple of times whether my pi is throttled, as I had no peripherals connected to it - it was okay.

It's a shame my pi died, I had it for like 4 days, even started doing project on it. Do you think that it's refundable? I'm sure I haven't short-circuited anything, the fan I've connected shouldn't cause overload and even if so - there's a polyfuse. The only thing that caused it seems to be the code snippet I've sent yesterday.

gabzim, 2019-05-17:
@drahoslove is there risk of ruining my pi by using this library? maybe that should be added as a disclaimer. what exactly should I avoid using from this lib?

shanghuiyang, 2019-10-25:
not all pins support pwm. please try pin 12, 18, 13 or 19, and need to use `sudo` to run your app.
pwm works well with pin 13 on my pi 3.