stianeikeland/go-rpio #44: Never any install documentation??????

momiccioli, 2019-04-14:
How do you install this library on the PI???????

xiegeo, 2019-04-16:
What do you mean by "install"?

momiccioli, 2019-04-16:
How do I add this library to my GO installation so I can compile with it.  There was a utility for OSX that took care of this.  I would like to compile directly on the Pi.

xiegeo, 2019-04-16:
This library is `go get`able. Just import according to the readme file. If
you run into any errors, report that.

You only have to install the go development tools, you don't install go
libraries.

momiccioli, 2019-04-18:
I understand that I use the import statement to pull the library into the go program.  However, there is nothing to tell where the library should be located on my hard drive.  I have tried different locations based on the GOLANG install and the relative PATH but nothing has worked.  People that are new to GOLANG may struggle to figure this out.  I have been in IT for over 30 years.  I have written code on Mainframes to PIs and almost everything in between.  So I am not a noob.  But there is a piece of information missing to get this right.

 

As I said on my MAC there was a script that took care of this and it worked on the first try.  But not on the Pi.  I find this very frustrating with people developing and posting. They tend to provide “some” use instructions but not deployment instructions.  I don’t mean to complain, hopefully you see my point.  So my question is….

 

How do I deploy the library to my GOLANG installation on the Pi so that I can compile with it.  You should include this information in your readme.

xiegeo, 2019-04-18:
That's part of the Go tooling, it depends on the version and environment
settings that you use. Every Go project is the same.

If it doesn't work out of the box, you system is miss configured. Instead
of telling me how you think it should work, tell me what errors you are
seeing when you try to build your project.

momiccioli, 2019-04-18:
Not telling you how it should work just how to ensure it can be used.  Before you ask your library is the only thing that does not work everything else is fine.

 

Here is the compile statement and the error

 

pi@octomax:~$ go install $GOPATH/gpio.go

Golang/src/gpio.go:7:2: cannot find package "github.com/stianeikeland/go-rpio" in any of:

        /usr/local/go/src/github.com/stianeikeland/go-rpio (from $GOROOT)

        /home/pi/Golang/src/src/github.com/stianeikeland/go-rpio (from $GOPATH)

 

pi@octomax:~$ echo $GOPATH

/home/pi/Golang/src

 

pi@octomax:~$ echo $GOBIN

/home/pi/Golang/bin

 

pi@octomax:~$ echo $GOROOT

/usr/local/go

 

pi@octomax:~$ ls -l /home/pi/Golang/src/github.com/stianeikeland/

total 80

drwxr-xr-x 4 pi pi  4096 Apr 14 10:15 go-rpio

-rwx------ 1 pi pi 75472 Apr 14 10:15 go-rpio.a

 

pi@octomax:~$ ls -l /usr/local/go/src/github.com/stianeikeland/

total 80

drwxrwxrwx 4 root root  4096 Apr 14 10:15 go-rpio

-rwxrw-rw- 1 root root 75472 Apr 14 10:15 go-rpio.a

 

All the files are where the compiler is looking.

xiegeo, 2019-04-18:
`pi@octomax:~$ go install $GOPATH/gpio.go` That's not how anything works.

And how did you get .a files in your src path? I would just nuke everything
and restart.

` /home/pi/Golang/src/src/github.com/stianeikeland/go-rpio (from $GOPATH)`
Here is the problem: src/src/. Your gopath should not end in src, otherwise
it add another src to it.

Forget everything you knew about linking to libraries and let go tooling do
it's magic. https://golang.org/doc/articles/go_command.html#tmp_3

tkuehn1, 2020-02-20:
If you want to install third party packages you have to enter a console and type go get github.com/stianeikeland/go-rpio

Go get will automatically include the package and you can use it in your Programm