stianeikeland/go-rpio #47: traffic light test faild on RPi 4

kadisi, 2019-09-10:
go version:  `go version go1.12.9 linux/arm`

`export GO111MODULE=on`

go code:
```
import (
        "log"
        "time"

        rpio "github.com/stianeikeland/go-rpio"
)

func main() {

        err := rpio.Open()
        if err != nil {
                log.Fatalf("open rpio error %v", err)
        }
        defer rpio.Close()

        red := rpio.Pin(13)
        red.Output()

        red.Low()

        for {
                red.High()
                time.Sleep(time.Second)
                red.Low()
                time.Sleep(time.Second)
        }
}
```

we i run my binary, the red light didn't respond.

pinout
```
,--------------------------------.
| oooooooooooooooooooo J8   +======
| 1ooooooooooooooooooo  PoE |   Net
|  Wi                    oo +======
|  Fi  Pi Model 4B  V1.1 oo      |
|        ,----.               +====
| |D|    |SoC |               |USB3
| |S|    |    |               +====
| |I|    `----'                  |
|                   |C|       +====
|                   |S|       |USB2
| pwr   |HD|   |HD| |I||A|    +====
`-| |---|MI|---|MI|----|V|-------'

Revision           : b03111
SoC                : BCM2711
RAM                : 2048Mb
Storage            : MicroSD
USB ports          : 4 (excluding power)
Ethernet ports     : 1
Wi-fi              : True
Bluetooth          : True
Camera ports (CSI) : 1
Display ports (DSI): 1

J8:
   3V3  (1) (2)  5V
 GPIO2  (3) (4)  5V
 GPIO3  (5) (6)  GND
 GPIO4  (7) (8)  GPIO14
   GND  (9) (10) GPIO15
GPIO17 (11) (12) GPIO18
GPIO27 (13) (14) GND
GPIO22 (15) (16) GPIO23
   3V3 (17) (18) GPIO24
GPIO10 (19) (20) GND
 GPIO9 (21) (22) GPIO25
GPIO11 (23) (24) GPIO8
   GND (25) (26) GPIO7
 GPIO0 (27) (28) GPIO1
 GPIO5 (29) (30) GND
 GPIO6 (31) (32) GPIO12
GPIO13 (33) (34) GND
GPIO19 (35) (36) GPIO16
GPIO26 (37) (38) GPIO20
   GND (39) (40) GPIO21

For further information, please refer to https://pinout.xyz/

```

and when i use python to test , it works.

```
#!/usr/bin/env python

import RPi.GPIO as GPIO
import time

RED = 13

GPIO.setmode(GPIO.BCM)
GPIO.setwarnings(False)
GPIO.setup(RED, GPIO.OUT)

def Close():
        GPIO.output(RED,False)

try:
        while(1):
            GPIO.output(RED,True)
            time.sleep(1)
            GPIO.output(RED,False)
            time.sleep(1)
except KeyboardInterrupt:
        Close()
        pass

```

kadisi, 2019-09-10:
ping @stianeikeland

kadisi, 2019-09-10:
ping @drahoslove

drahoslove, 2019-09-10:
@kadisi Try to look again at your logic, especially the part inside your loop ;)

Hint 1: it's not the same as in your python code.

Hint 2: toggle

kadisi, 2019-09-10:
> Try to look again at your logic, especially the part inside your loop ;)
> Hint 1: it's not the same as in your python code.
> Hint 2: toggle

@drahoslove I don't think this is the reason.

Is it a problem with bcm?

my bcm is bcm2711  while go-rpio support bcm2835.

https://github.com/stianeikeland/go-rpio/blob/a36b96d0b1a40b37017d64875275e7e6312408c7/rpio.go#L32

drahoslove, 2019-09-10:
Ah sorry, was too sleepy while responding before, now I see,...

You should use this pin: https://pinout.xyz/pinout/pin33_gpio13

But I'm not sure if bcm2711 is compatible with bcm2835. Maybe it's not.

kadisi, 2019-09-10:
@drahoslove  thanks for responding

i did use `13` in my code.

```
        red := rpio.Pin(13)
        red.Output()

```

do you test on RPi 4?

wfd3, 2019-10-23:
This is, I think, due to the Pi 4 having a different GPIO base address than the 3.  Moreover, the address is at a different offset in /proc/device-tree/soc/ranges.

The Pi3 has a base address of 0x3f00000, which can be found at offset 4 in /proc/device-tree/soc/ranges.  The Pi4 has a base address of 0xfe000000, which is at offset 8.  

If you override the return value in getBase(), it works (at least on my Pi 4).  I'll take a pass at an update to getBase() but it'll take me a while to get to it.

kadisi, 2019-10-23:
@wfd3 thanks you.

wfd3, 2019-10-23:
Turned out it was easier than I thought. Try this patch.  I've tested it on my Pi4 and Pi3 and it seems to work.  I can't speak to any other Pi variants.

`--- a/rpio.go
+++ b/rpio.go
@@ -714,23 +714,38 @@ func Close() error {
 
 // Read /proc/device-tree/soc/ranges and determine the base address.
 // Use the default Raspberry Pi 1 base address if this fails.
-func getBase() (base int64) {
-       base = bcm2835Base
+func readBase(offset int64) (int64, error) {
        ranges, err := os.Open("/proc/device-tree/soc/ranges")
        defer ranges.Close()
        if err != nil {
-               return
+               return 0, err
        }
        b := make([]byte, 4)
-       n, err := ranges.ReadAt(b, 4)
+       n, err := ranges.ReadAt(b, offset)
        if n != 4 || err != nil {
-               return
+               return 0, err
        }
        buf := bytes.NewReader(b)
        var out uint32
        err = binary.Read(buf, binary.BigEndian, &out)
        if err != nil {
-               return
+               return 0, err
+       }
+       return int64(out), nil
+}
+
+func getBase() int64 {
+       // Pi 2 & 3 GPIO base address is at offset 4
+       b, err := readBase(4)
+       if err == nil && b != 0 {
+               return b
        }
-       return int64(out)
+       // Pi 4 GPIO base address is as offset 8
+       b, err = readBase(8)
+       if err == nil && b != 0 {
+               return b
+       }
+       // Default to Pi 1
+       return int64(bcm2835Base)
+
 }
`

wfd3, 2019-10-23:
Turns out it was easier than I thought.  Try this patch.  I've tested it on my Pi3 and Pi4, but I can't speak to other Pi variants. 
[rpio.patch.txt](https://github.com/stianeikeland/go-rpio/files/3762650/rpio.patch.txt)

drahoslove, 2019-10-23:
@wfd3 This looks good.
Could you please create a pull request with this patch, after kadisi confirms it works for him? 
I don't have a Pi 4 to test it.

kadisi, 2019-10-24:
@wfd3  i have testd, it works.    I am looking forward to your pr。

@drahoslove thank you

wfd3, 2019-10-24:
Pull Request is in the queue.