stianeikeland/go-rpio #52: Rotary Encoder support?

djensenius, 2019-12-12:
Are there any plans for rotary encoder support?

drahoslove, 2019-12-16:
@djensenius What kind of support would you like to see?

I believe you might be already able to use what `go-rpio` offers you to support your rotary encoder (with your custom implementation).

Does `go-rpio` lack any crucial feature which prevents you from doing so? If so please specify.

Anyway, any specific implementation to support any specific peripheral device would be too specific to be part of `go-rpio`. We might try to support some general interfaces, like SPI, I2C etc..  but not specific devices - that is up to the users.

djensenius, 2019-12-27:
Thanks @drahoslove - this makes sense!

closed 2019-12-27 by djensenius