stianeikeland/go-rpio #54: Unable To Find Compatible WiringPi Pinouts With This Library

bonedaddy, 2020-01-25:
I'm currently trying to write a golang equivalent of [this](https://github.com/dragino/rpi-lora-tranceiver/blob/master/dragino_lora_app/main.c) Raspberry Pi LoRa hot module, which uses wiringPi pniouts.

I've tried the [wiringPi pin out numbers](https://github.com/dragino/rpi-lora-tranceiver/blob/master/dragino_lora_app/main.c#L166-L168) in [this](https://github.com/RTradeLtd/rpi-lora-tranceiver/blob/master/src/lora.go) golang library using `go-rpio` and I'm unable to get it to work properly, and it fails with a `unrecognized traceiver` error.

# System Info

Model: Raspberry Pi 3B+
```
No LSB modules are available.
Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 9.11 (stretch)
Release:	9.11
Codename:	stretch
```

```
Linux rpi3 4.19.66-v7+ #1253 SMP Thu Aug 15 11:49:46 BST 2019 armv7l GNU/Linux
```

bonedaddy, 2020-01-26:
Small update that I forgot to post yesterday, the manufacturer of the board I'm using (dragino LoRa/GPS HAT), has this diagram http://wiki.dragino.com/images/9/9b/RPi_GPIO.jpg.

So based off the pinouts that the C code is using, the translation to BCM numbers should be:

```
int ssPin = 25;
int dio0  = 4;
int RST   = 17;
```

But unfortunately `go-rpio` still didn't work

drahoslove, 2020-02-14:
You are right, you have to use the `BCM` pin numbering. If it does not work, the issue is probably somewhere else.

I am sorry I think I am not able to help you.

drahoslove, 2020-02-14:
I noticed some differences between the main.c and your  lora.go:
https://github.com/RTradeLtd/rpi-lora-tranceiver/blob/master/src/lora.go#L46 . - you are using `RST` here instead of `ssPin`

https://github.com/RTradeLtd/rpi-lora-tranceiver/blob/master/src/lora.go#L65 - this should be input

Some ideas: 
 - try to change  `SpiChipSelectPolarity` and/or `SpiMode`
 - try to add some sleep after SpiExchange 
 - you have to run the program as rood for spi functions to work

bonedaddy, 2020-02-14:
Hello, thanks for taking the time to look at the code in detail like that.

I will try your suggestions and hopefully they work!