stianeikeland/go-rpio #61: strangely, does *not* work with sudo

dubaxx, 2020-04-28:
I'm having a perplexing issue here, hopefully someone can shed some light on the subject.

go 1.14, raspberry pi 4 B+

using the blinker.go example:

./blinker works just fine
sudo ./blinker makes nothing happen.

Why is this?  Normally that would be fine, except that I'm trying to use rpio along with a dht11 library that requires root access to run.  I find it perplexing that running rpio with heightened privileges causes it to _not_ work.

srmihalampret, 2020-05-09:
Same here - had to switch from `root` user to `pi` user to get blinker.go working. Not sure what is the reason for this.

mlampret, 2020-05-09:
Same here - had to switch from `root` user to `pi` user to get blinker.go working. Not sure what is the reason for this.

openwrd, 2020-05-21:
I also encountered the same problem.

drahoslove, 2020-06-06:
This might be some OS-related change.
You can try to change this part of code: 
https://github.com/stianeikeland/go-rpio/blob/master/rpio.go#L630-L637

So that `/dev/gpiomem` is always used (but only basic gpio functionality will work)

drahoslove, 2020-06-06:
Also #50 this might solve your issue, can you please try?

s00500, 2020-12-19:
Same issue here... can not get it to work on root on and RPI4, go 1.15.3
Only trying to control a single pin at the moment, program works with pi

rami-dabain, 2020-12-31:
related https://github.com/stianeikeland/go-rpio/issues/68 [fixed]

drahoslove, 2021-08-19:
Should be fixed now - thanks to #70 by @rami-dabain.

Feel free to reopen or create a new issue in case of a problem.

closed 2021-08-19 by drahoslove