stianeikeland/go-rpio #64: Nice library but seems PWM is behaving a bit strangely

hdahlgren, 2020-08-17:
Perhaps it's just me but whenever I try to use PWM the BCM pin 2 changes from ALT0 to ALT2 and suddenly it's no longer possible to use ADC chips to for example read the value of a Rotary Potentiometer and then controlling a DC motor using PWM. Doesn't seem to matter if I use PWM on BCM pin 12 or BCM 13.

Using Raspberry PI 4.

mjgaga, 2020-10-11:
Same problem

drahoslove, 2021-08-19:
That's interesting behavior. Did you guys find out which function call was causing it?

Also, it might be fixed by #50, which was in master for a long time, but unfortunately, it wasn't tagged so it was not used if you were using go modules. I created the new tag now, so you might try to update it (`go get -u github.com/stianeikeland/go-rpio/v4`) and check whether it solves your issue.