stianeikeland/go-rpio #65: Dockerize?

vertoforce, 2020-08-25:
Anybody ever dockerize an app using this library on a raspberry pi?  Is it as simple as just making a volume for `/dev/mem`?

vertoforce, 2020-08-27:
For anyone looking to do this, check [this out](https://stackoverflow.com/questions/48441737/docker-error-no-access-to-dev-mem-try-running-as-root).

slimcdk, 2020-10-30:
Hi! How did you get this to work? 

I'm using the simple blinker example which works fine directly on Raspberry Pi OS, but not inside the container. I see the logs I've added, so I know it should be controlling GPIO.

I've tried a bunch of combinations of 
`--privileged`
`--device=/dev/mem`
`--device=/dev/gpiomem`
` -v /dev/mem:/dev/mem`
` -v /dev/gpiomem:/dev/gpiomem`

but none of them works.

Thanks!