stianeikeland/go-rpio #68: RPI 4 B PWM not working

ScottDLance, 2020-10-20:
I know this has been mentioned in several other issues but I am putting it here for a specific issue. 

I'm running on a RPI 4 Model B.   Running output on Pin 18 and setting the pin to High works.  Setting the same Pin 18 to PWM and setting it to a 100% duty cycle does not work.

I'm running the program as `root`.

There has been suggestions made that the PWM functionality was fixed for the RPI 4.  This does not appear to be the case.

ntavelis, 2020-11-07:
Hey @ScottDLance, It works for me. I am on raspberry pi 4 B.

Note, it only works with sudo privileges

rami-dabain, 2020-12-31:
Yup. same here, sudo doesn't work.  ntavelis care to share a sample that works for you? Keep in mind that some kernel changes were done in raspian, maybe if you upgrade it will stop working as well

ntavelis, 2020-12-31:
Hi, the below code works for me: 

```
package main

import (
        "os"
        "time"
        "github.com/stianeikeland/go-rpio"
)

func main() {
        err := rpio.Open()
        if err != nil {
                os.Exit(1)
        }
        defer rpio.Close()

        pin := rpio.Pin(18)
        pin.Mode(rpio.Pwm)
        pin.Freq(64000)
        pin.DutyCycle(0, 32)
        // the LED will be blinking at 2000Hz
        // (source frequency divided by cycle length => 64000/32 = 2000)

        // five times smoothly fade in and out
        for i := 0; i < 5; i++ {
                for i := uint32(0); i < 32; i++ { // increasing brightness
                        pin.DutyCycle(i, 32)
                        time.Sleep(time.Second/32)
                }
                for i := uint32(32); i > 0; i-- { // decreasing brightness
                        pin.DutyCycle(i, 32)
                        time.Sleep(time.Second/32)
                }
        }
}
```

I have noticed something interesting, when I use go modules, the above sample does not work. When I do not use go modules the above sample works. Not sure go modules affects the behavior, maybe something related with the paths inside the package. Just though to share it.

rami-dabain, 2020-12-31:
this is the official example, yes it does not work for me. Interesting note about the paths. will check this!

rami-dabain, 2020-12-31:
got it fixed in https://github.com/stianeikeland/go-rpio/pull/70 !!!
It was a tiny permission bug

baumgarb, 2021-05-19:
> 
> I have noticed something interesting, when I use go modules, the above sample does not work. When I do not use go modules the above sample works. Not sure how go modules affect the behavior, maybe something related with the paths inside the package. Just though to share it.

That's indeed interesting. I'm experiencing the same issue. If I'm using go modules the code above is not working. If I move all the code to my GOPATH and stop working with go modules it works (given that I run it as root, of course).

Peelz, 2021-08-12:
I'm facing same issue, work around by build binary and exec with run.

drahoslove, 2021-08-19:
Should be fixed now - thanks to #70 by rami-dabian.

Feel free to reopen or create a new issue in case of a problem.

closed 2021-08-19 by drahoslove