stianeikeland/go-rpio #71: pullup does not work on RPI 4 B

wenwenchenbosch, 2021-01-11:
I tried to build and run the pullup example https://github.com/stianeikeland/go-rpio/blob/master/examples/pullup/pullup.go. But it does not build, see error message:

```
pi@raspberrypi ~/g/e/pullup> go build
# github.com/stianeikeland/go-rpio/v4/examples/pullup
./pullup.go:26:48: pin.ReadPull undefined (type rpio.Pin has no field or method ReadPull)
./pullup.go:30:50: pin.ReadPull undefined (type rpio.Pin has no field or method ReadPull)
note: module requires Go 1.13~
```
After I removed the `ReadPull()` call, I got the following output:

```
pi@raspberrypi ~/g/e/pullup> ./pullup
PullUp: 0, 0
PullDown: 0, 0
```

That means PullUp does not work.

fdidron, 2021-02-03:
@wenwenchenbosch I think this is because the latest tag (v4.4.0) is not pointing to the master branch.
@stianeikeland  is it possible to tag a new version ?

fdidron, 2021-02-03:
@wenwenchenbosch  as a workaround what I did was

- `go get github.com/stianeikeland/go-rpio` outside of my project (so it doesn't touch my project's go mod file)
- Add  `replace github.com/stianeikeland/go-rpio/v4 v4.4.0 => ../../stianeikeland/go-rpi` at the bottom of my go.mod project file
- Run `go mod tidy` at the root of my project

jkandasa, 2021-03-20:
to get the master branch: `go get github.com/stianeikeland/go-rpio/v4@master`

drahoslove, 2021-08-19:
I fixed the examples so they use the modules version 4 now, and I pushed a new tag.
It should solve this issue. Feel free to reopen it if it does not.

(also in similar situations it might help to try to delete go mod cache first by `go clean --modcache`)

closed 2021-08-19 by drahoslove