stianeikeland/go-rpio #82: Hardlock when EdgeDetect is used

jax-b, 2022-01-16:
I found a small issue with the library. For some reason when rpio is close and EdgeDetect is still enable it will hard lock the rpi. i tested it on both a pi3 and pi4. if i call pin.Detect(rpio.NoEdge) before close nothing locks up and everything is fine. I don't have any logs unfortunately dmsg and the kernel log give no insight.

go version: go1.17 linux/arm64
kernel version (uname-a):   Linux XXXX 5.13.0-1013-raspi #15-Ubuntu SMP PREEMPT Fri Jan 7 23:16:57 UTC 2022 aarch64 aarch64 aarch64 GNU/Linux
go-rpio version: 4.5.1

yzj578492228, 2022-01-16:
这是来自QQ邮箱的假期自动回复邮件。
 
您好，我最近正在休假中，无法亲自回复您的邮件。我将在假期结束后，尽快给您回复。

jax-b, 2022-01-16:
I have updated to 4.6 and it is still a issue, I had one run where instead of crashing it locked the file system

rspier, 2023-01-05:
Found 
> WARNING: this might make your Pi unresponsive, if this happens, you should either run the code as root, or add `dtoverlay=gpio-no-irq` to `/boot/config.txt` and restart your pi,

in the docs.

  It came from https://github.com/stianeikeland/go-rpio/issues/35, of which this bug is likely a dupe.

peritonow, 2023-10-17:
yes,' dtoverlay=gpio-no-irq to /boot/config.txt' was the solution in my case. Thank you.