stianeikeland/go-rpio #88: Interrupt/event based edge detection

wjessop, 2023-09-06:
I'm trying to detect a pulse on a GPIO pin (the tacho from a fan). Right now I have to poll, something like this:

```go
for {
  if rpmPin.EdgeDetected() {
    doTheThint()
  }
}
```

This busy loop takes 100% of a CPU core. I can add a sleep in there but at 10k pulses per minute the sleep has to be small and the program still uses a lot of CPU. Would it be possible to use some sort of hardware based interrupt notification for edge detection to ease the load on the CPU?

peritonow, 2023-10-17:
this information/link might be useful https://www.i-programmer.info/programming/hardware/14114-raspberry-pi-iot-in-c-events-a-interrupts.html?start=1

decadenza, 2025-01-03:
Bump.
I've also seen the comment https://github.com/stianeikeland/go-rpio/blob/939e5aba54aa8c891539f3f1187d7e20431390d8/rpio.go#L481

As of today, what is the best way to avoid polling?
Thanks.

wjessop, 2025-01-03:
I don't have an answer for you unfortunately @decadenza, this was a toy project I haven't touched since pretty much the initial issue. I did end up getting the fan RPM monitoring working, but I can't remember if I fixed the high CPU issue or not. Here's the code, which may or may not work for you. It was a work in progress so contains some commented out stuff and I can't remember if it's all relevant or not, but it might help:

```go
// MeasureFanRPM starts background processes for gathering the fan RPM data
func (rpm *FanRPMMonitor) MeasureFanRPM() {
	rpmPin := rpio.Pin(rpm.rpmMonitorPin)
	rpmPin.Input()
	rpmPin.PullUp()
	/*
		This next line of code will crash the Pi unless you have disabled interrupts for GPIO pins:

					Add dtoverlay=gpio-no-irq to /boot/config.txt and restart your pi.

					Explanation:
					Thanks, roffe, for the example - finally something I could reproduce.
					I realized that the difference between the unit test which works fine and the cases where something useful is connected to pin is that the number of edges generated is much higher than the number of calls to EdgeDetected(). (Single button press generates many edges too due to "switch bounce effect".)

					Every edge event generates interruption as the side effect, but these are not handled by the system and after some number of unhandled interruptions it freezes.

					In some update of Linux kernel (to which I switched after implementing the event detection functionality) the way the interruptions are handled changed. It worked before "by accident".
					More info here: raspberrypi/linux#2550 (comment)

					The solution above will disable all gpio interruption requests. But it is just workaround since it might break some other programs. Therefore I'll try to figure out a better solution. Either by disabling irq from within go-rpio, or better by reimplementing the detection functionality and actually handle the interruptions.

					But disabling irq using dtoverlay in config should be good enough for now.

					https://github.com/stianeikeland/go-rpio/issues/35
	*/
	rpmPin.Detect(rpio.RiseEdge)

	// Attempt to use a new lib that will use interrupts to handleedge
	// detection. It doesn't work, but there's an open issue. It may be worth
	// re-visiting in the future if it gets fixed.
	//
	// https://github.com/brian-armstrong/gpio/issues/17
	//
	// watcher := gpio.NewWatcher()
	// watcher.AddPinWithEdgeAndLogic(rpmMonitorPin, gpio.EdgeRising, gpio.ActiveHigh) defer watcher.Close()

	go func() {
		for {
			select {
			case <-rpm.fanRPMTimeout.C:
				rpm.FanRPM = 0
			default:
				if rpmPin.EdgeDetected() {
					rpm.rpmCounter()
				}
			}

			// Try an interrupt based approach
			// _, value := watcher.Watch()
			// if value == 1 {
			// 	// The pin went high, we don't care about pins going low
			// 	rpmCounter()
			// }

			/*
				By my calculations (confirmed by oscilloscope) there is a maxuimum of 1 / ((5000 * 2) / 60 ) =
				0.006s (6 milliseconds) between possible edges:

				- 5000 RPM
				- 2 pulses per rotation

				So, sleeping for 4 milliseconds should be fine here as the goroutine
				should always be able to wake up in time for the next pulse.

				HOWEVER sleeping for this long causes incorrct readings which is
				great. I'm disabling RPM detection for now so it doesn't matter for
				now. Anyway, even a 4ms delay causes over 15% CPU use at times.
			*/
			time.Sleep(4 * time.Millisecond)
		}
	}()
}
```

let me know if you find a solution!

decadenza, 2025-01-06:
Thanks @wjessop , pretty good for a toy project!

It froze for me, despite disabling irq using dtoverlay in `boot/config.txt` and rebooting.

A straightforward library like this will be useful as I cannot find a way to manage interrupts from Go (and polling is horrible). I tried others that did not work either.

I'd also love to help, but honestly I don't know where to start.

wjessop, 2025-01-06:
Re-reading the code it looks like I never did solve the issue, I added in a sleep which meant that the CPU was only pinned at 15%. Good luck!